lnlp/LMIC-node #10: Non-standard config for Feather M0

snoopen, 2021-05-12:
Line 88 of `adafruit_feather_m0_lora.h` has dio1 on pin 5. However the common pin used on all examples I've seen uses pin 6.

Additionally, and I'm not sure if it makes a difference, but the examples also has `.rssi_cal = 8,` instead of 10.

lnlp, 2021-05-12:
Thanks for your feedback.

I must have forgotten to change it in the actual code (after changing it from 5 to 6 in the documentation).
_Not having the hardware to test makes this easier to slip through._

> Additionally, and I'm not sure if it makes a difference, but the examples also has `.rssi_cal = 8,` instead of 10.

Ah, I see, the 'feather-us915' examples use 8, while file `hal/getpinmap_featherm0lora.cpp` uses 10.
The 8 is possibly more correct. I'm not sure where you would see the actual difference. I suspect in reported RSSI value of downlink messages.

lnlp, 2021-05-13:
DIO1 mapping has been fixed.

Will have a look at `rssi_cal` later.

lnlp, 2021-05-15:
rssi_cal has been changed to 8.

closed 2021-05-15 by lnlp

snoopen, 2021-05-21:
Awesome thank you!

My initial testing with these settings got the device working with the expected serial monitor output. Tho I haven't been able to test on TTN as there are no active gateways near enough to me.