lnlp/LMIC-node #11: Limit SF range for auto configuration

udovic2100, 2021-06-08:
I really like this example: very well documented and easy setup.
I would like to change the SFs that the auto configurations uses. I am quite far away from the next gateways and only messages with SF > 9 will reach the next gateway. I there a way to change the SF range from 7 to 12 to 10 to 12?

lnlp, 2021-06-08:
Thanks for your compliments.

Yes it is possible to set SF manually but its effect depend on several factors.

When using OTAA activation, you can initially set the SF but it will be used for the join only. During the join the SF will be updated by the network server automatically with a SF that appears suitable to the networkserver.

When using ABP activation there is no join and the SF will not be changed. If ADR is enabled then SF may be changed over time (IIRC after the first 20 uplinks) by ADR MAC downlink command from the network server. In case of ADR the network server will select a SF that appears most suitable for the node and its signal quality (which e.g. depends on distance to gateways).

It is preferred to use OTAA for multiple reasons.

Use ADR only for stationary nodes (end devices) but not for mobile devices that are on the go.

SF is normally set automatically but if you want to set a value explicitly this is can be done with the LMIC API function `LMIC_setDrTxPow(dataRate, txPower)`. This is a combined function that sets both data rate and transmit power.

In LMIC-node SF can be specified in the `initLmic()` call in setup (so you don't have to call `LMIC_setDrTxPow()` explicitly).
When using ABP it suffices to only specify adrEnabled and dataRate (e.g. DR_SF9 while DR_SF7 is the default).
When using OTAA you will have to specify all 4 parameters when calling `initLmic()` because it will only use the dataRate and txPower parameters for OTAA if `setDrTxForOtaaExplicit` is set to true.

`initLmic()` signature:

```cpp
void initLmic(bit_t adrEnabled = 1, 
              dr_t dataRate = DR_SF7, 
              s1_t txPower = 14, 
              bool setDrTxPowForOtaaExplicit = false) 
```

Assuming that you would use ADR, to use SF9 you would have to use the following `initLmic()` call in `setup()`:
_(To disable ADR the first parameter should be 0.)_

For OTAA:

```cpp
initLmic(1, DR_SF9, 14, true)    // DR_SF9 will only be used for the initial join
```

For ABP:

```cpp
initLmic(1, DR_SF9)
```

udovic2100, 2021-06-08:
Thanks for the really quick answer. I'm using OTAA and changed the initLmic() as you suggested in the setup. However I get the following terminal output: 

```
LMIC-node

Device-id:     rpi-pico
LMIC library:  MCCI
Activation:    OTAA
LMIC debug:    2
Interval:      60 seconds

RXMODE_RSSI
000000184937:  Event: EV_JOINING
184990: engineUpdate, opmode=0x4

000000185097:  doWork job started
261466: engineUpdate, opmode=0x4
000000261593:  Event: EV_TXSTART
261684: TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5, IH=0
577299: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 3
578011: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 578459 rxtime: 577924 entry-rxtime: 535 now-entry: 4 rxtime-txend: 312375
639799: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
640507: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 654782 rxtime: 640424 entry-rxtime: 14358 now-entry: 4 rxtime-txend: 374875
655034: processRx2Jacc txrxFlags 0x2 --> 00
000000655156:  Event: EV_JOIN_TXCOMPLETE
```

Correct me if I'm wrong, but it looks as if my change didn't have any effect on the joining process. I was hoping for SF=10 on the first try.

Thank you for your help in advance.

lnlp, 2021-06-09:
You're right. That shows SF7 and not SF10.

Partially I'm to blame for that because I forgot that LMIC-node (currently) only calls `LMIC_setDrTxpow()` for OTAA is ADR is **disabled**.

I just did some tests with a slightly modified version of LMIC-node where the ADR check is removed, but unfortunately that does not make any difference. The reason for this behavior is determined by the LMIC library, not by LMIC-node.
I therefore am unable to make any changes to LMIC-node that would allow to set a custom SF value used for the join.

This matter is not simple because the initial SF used for a join is LoRaWAN region dependent and so is the (max) txPower value. Also there is no API to set SF and txPower independently. For additional information see the following related [discussion](https://forum.mcci.io/t/lmic-setdrtxpow-questions/99).

So currently it is not possible to set a custom SF for a join.

udovic2100, 2021-06-09:
Thank you very much for your explanation. I follow the discussion.

closed 2021-06-10 by lnlp