lnlp/LMIC-node #14: Question on approach to use ESP32 Devboards and RFM95

peterl3233, 2021-07-10:
This code works very well (indeed, thanks) on Heltec but I can't make it work on a generic DevKitC-v1 ESP32 with external RFM95.  My approach has been to adjust a Lolin-d32 file and get the pin mappings right - so far without success.

Has anyone used this code on ESP32/RFM95 or can you offer an approach to add ESP32 dev module to the list of configured devices?

lnlp, 2021-07-12:
> My approach has been to adjust a Lolin-d32 file and get the pin mappings right

I assume you mean the BSF file here?
**What did you adjust and why?** What is not working, do you get any errors?

I don't have an Espressif  DevKitC board to test but as far as I can see you should be able to use either lolin32 or lolin_32 as basis.

I2C and SPI pins for these boards are the same. The Lolin boards do have onboard user LED connected to GPIO5 though which in the Arduino BSP is defined SS (SPI chip select, SS/NSS/CS), which is why I used different GPIO's for SPI chip select for these boards.

The DevKitC does not have onboard user LED so you could use the default GPIO5 ('SS') for SPI chip select for the DevKitC as well. However, with the pins defined in the BSF's for lolin32 and lolin_d32 the DevKitC should work just fine as far as I can see.

What I would like to suggest (and possibly could make a difference) is change the value `board` parameter (in the board specific section) in platformio.ini from `lolin_d32` (or `lolin32`, whichever you want to use and modify) to `esp32dev`.

Another option may be to change the SPI frequency in the BSF from 8 MHz to 1 MHz, depending on the quality of your wiring/breadboard.

Also disable USE_LED in the board specific section in platformio.ini because the DevKitC board does not have an onboard user LED.

If this does not help you make have to double-check your physical wiring, pin mappings defined in the BSF and whether you have actually selected the proper board-id in platformio.ini (so it uses the BSF that you are assuming).

> can you offer an approach to add ESP32 dev module to the list of configured devices?

I could add support for DevKitC but would follow similar steps as described above to add it, so you should at least be able to get your board working with the current version of LMIC-node.

To add it I would need some more information and specifications (e.g. correct full name) though.
I assume that you DevKitC-V1 is an "Espressif DevKitC-V1" board and not a DoIt DevKit V1?
Unfortunately, on the Espressif site I only see reference design documentation for ESP32-DevKitC-V2 and ESP32-DevKitC-V4.

lnlp, 2021-07-16:
@peterl3233  Please respond.

lnlp, 2021-07-19:
@peterl3233 No response, so topic closed.

closed 2021-07-19 by lnlp

peterl3233, 2021-07-27:
Hi .. Thanks for reply ... I'll look at all this carefully but got sidetracked with a family matter.  I appreciate your response very much.

lnlp, 2021-07-28:
Thanks. If you find some time and want to continue here, just let me know and I can reopen the issue.

peterl3233, 2021-08-12:
> @peterl3233 Did you succeed to get it working?

Hi ... FYI, following your suggestions, I did get this working.  Your code was without error and after I checked all the points you made I dug deeper and eventually found a "flakey" RFM95 module; so it was actually a hardware problem.  After replacing teh RFM95, I use it with ESP32 with RMM95 with an SHT30 sensor on I2c.  Works great on Chirpstack .. I didn't try on TTN.

For interest, here is the pinmap for the Espressive ESP32 Devkit_v4 as connected to RFM95 module:
```
const lmic_pinmap lmic_pins = {
    .nss = 5,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 14,
    .dio = { /*dio0*/ 26, /*dio1*/ 33, /*dio2*/ LMIC_UNUSED_PIN }
#ifdef MCCI_LMIC
    ,
    .rxtx_rx_active = 0,
    .rssi_cal = 10,
    .spi_freq = 8000000     /* 8 MHz */
#endif    
};
```
I created a "bsf_ESP32_RFM95.h" file and if you would like, I'll send it to you or post it on github.  Below is the changed (hardly) code block for ESP32_RFM95:

 ```
[env:ESP32_RFM95]
; Espressif ESP32 DevKit_v4.
; No display.
; No LED.
platform = espressif32
board = ESP32_RFM95
framework = arduino
upload_speed = 921600
monitor_speed = ${common.monitor_speed}
lib_deps =
    ${common.lib_deps}    
    ${mcci_lmic.lib_deps}
build_flags =
    ${common.build_flags}
    ${mcci_lmic.build_flags} 
    -D BSFILE=\"boards/bsf_ESP32_RFM95.h\"
    -D MONITOR_SPEED=${common.monitor_speed}
    -D LMIC_PRINTF_TO=Serial    
    -D USE_SERIAL
    ; -D USE_LED       ; Not used
    ; -D USE_DISPLAY    ; Not used
```

nordblick2, 2024-02-15:
Ay chance to see that bsf_ESP32_RFM95.h content anywhere?😀