lnlp/LMIC-node #15: LMIC-node  project adding MCCI LoRaWAN LMIC library@^4.0.0

mitjad123, 2021-07-16:
Hi,

I am setting LMIC-node project on my win10 platformIO project targeting heltec wireless stick lite
I tried to add mcci-catena/MCCI LoRaWAN LMIC library@^4.0.0 library and got this error:

PIO Core Call Error: "Error: Invalid 'D:\\userX\\Documents\\PlatformIO\\Projects\\HelctecWiFiLORAone4all\\platformio.ini' (project configuration file): 'No section: 'common''"

My platforio.ini:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:heltec_wireless_stick_lite]
; Heltec Wireless Stick Lite (ESP32).
; No display.
platform = espressif32
board = heltec_wireless_stick_lite
framework = arduino
upload_speed = 921600
monitor_speed = ${common.monitor_speed}
lib_deps =
    ${common.lib_deps}    
    ${mcci_lmic.lib_deps}
build_flags =
    ${common.build_flags}
    ${mcci_lmic.build_flags} 
    -D BSFILE=\"boards/bsf_heltec_wireless_stick_lite.h\"
    -D MONITOR_SPEED=${common.monitor_speed}
    -D LMIC_PRINTF_TO=Serial    
    -D USE_SERIAL
    -D USE_LED
    ; -D USE_DISPLAY             ; Requires external I2C OLED display

mitjad123, 2021-07-16:
Aha, it looks like I would have to add MCCI lib before and changed platforio.ini

lib_deps =
    ${common.lib_deps}    
    ${mcci_lmic.lib_deps}
    mcci-catena/MCCI LoRaWAN LMIC library@^4.0.0

lnlp, 2021-07-16:
Apparently you have edited the `platformio.ini` file included with LMIC-node and have removed essential parts e.g. the [common] and other sections.
There is no need to remove anything from platformio.ini. Use the original `platformio.ini` and the problem should be solved.

If you wan't to slim down platformio.ini for just your board, be careful with what you change or remove and only do this if you understand the consequences.

mitjad123, 2021-07-19:
Hi, sorry for being negligent.
You were right. After using the original ini file, it almost worked.
I only had to change:
    -D BSFILE=\"boards/bsf_heltec_wireless_stick_lite.h\"
To:
    -D BSFILE=\"../boards/bsf_heltec_wireless_stick_lite.h\"

Because otherwise I had an error:
In file included from src\LMIC-node.cpp:53:0:
src\LMIC-node.h:79:46: fatal error: boards/bsf_heltec_wireless_stick_lite.h: No such file or directory
compilation terminated.
*** [.pio\build\heltec_wireless_stick_lite\src\LMIC-node.cpp.o] Error 1
================================================================================== [FAILED] Took 2.84 seconds =============================================================================

Thanks
Best regards

Mitja

lnlp, 2021-07-20:
Thanks for your feedback.

Regarding:

> I only had to change:
> -D BSFILE="boards/bsf_heltec_wireless_stick_lite.h"
> To:
> -D BSFILE="../boards/bsf_heltec_wireless_stick_lite.h"

That is strange because `boards` is a subfolder of `src`, they are not siblings (which the `../` implies).
Maybe you changed something to the folder structure?

mitjad123, 2021-07-20:
Yes, I did (as well as some other stupid things). But it is OK now. 
Thanks
Mtija

closed 2021-07-20 by mitjad123