lnlp/LMIC-node #17: Pro8MHz board does join but does not transfer messages

SensorsIot, 2021-07-22:
I wanted to use a LoRa Radio Node board with an 328 chip and OTA on the V3 stack. It uses the old LMIC library and joins, but does not transfer messages. Serial stops after: 

LMIC-node

Device-id:     pro-mini
LMIC library:  Classic [Deprecated]
Activation:    OTAA
Interval:      60 seconds

Clock Error:   30000 ppm (1965)    
000000000817:  Event: EV_JOINING   

000000000882:  doWork job started

Unfortunately, the memory is too small to use the new MCCI library. Was anybody successful to use an "Arduino-like" board on the V3 stack?

lnlp, 2021-07-22:
Hi A. @SensorsIot,

LMIC-node was successfully tested on the TTS CE V3 stack (aka TTN V3).
All supported boards were tested on both V3 and V2, the pro-mini included.
_(Only the Heltec Wireless Stick has not yet been tested, not by me at least.)_

For V3 OTAA device settings I used:
LoRaWAN version: MAC V1.0.3
Regional Parameters version: PHY V1.0.3 REV A
Frequency plan: EU863-870 MHz (SF9 for RX2 - recommended)
(and AppEUI: all 00's)

It works fine for my pro-mini. Below the serial output: _(I have enabled external display as well, so I can see output on both)_:
(downlinks work as well)

```text
LMIC-node

Device-id:     pro-mini  
LMIC library:  Classic [Deprecated]
Activation:    OTAA
Interval:      20 seconds

Clock Error:   30000 ppm (1965)
000000032827:  Event: EV_JOINING 

000000036598:  doWork job started
000000407266:  Event: EV_JOINED  

000000411115:  doWork job started
000000414311:  Input data collected
               COUNTER value: 1    
000000416032:  Packet queued
000000811667:  Event: EV_TXCOMPLETE
               Up: 1,  Down: 0

000001661121:  doWork job started
000001664315:  Input data collected
               COUNTER value: 2    
000001666037:  Packet queued
000002061759:  Event: EV_TXCOMPLETE
               Up: 2,  Down: 0

000002911129:  doWork job started
000002914322:  Input data collected
               COUNTER value: 3    
000002916045:  Packet queued
000003311959:  Event: EV_TXCOMPLETE
               Up: 3,  Down: 0

000004161137:  doWork job started
000004164330:  Input data collected
               COUNTER value: 4    
000004166054:  Packet queued
000004561793:  Event: EV_TXCOMPLETE
               Up: 4,  Down: 0

000005411143:  doWork job started
000005414338:  Input data collected
               COUNTER value: 5    
000005416061:  Packet queued
000005811807:  Event: EV_TXCOMPLETE
               Up: 5,  Down: 0

000006661152:  doWork job started
000006664344:  Input data collected
               COUNTER value: 6    
000006666066:  Packet queued
000006989141:  Event: EV_TXCOMPLETE
               Up: 6,  Down: 2
               Downlink received
               RSSI: -86 dBm,  SNR: 10.0 dB
               Port: 100
               Length: 1
               Data: C0
               Reset cmd received
000006999511:  Counter reset

000007911158:  doWork job started
000007914351:  Input data collected
               COUNTER value: 1    
000007916075:  Packet queued
000008311822:  Event: EV_TXCOMPLETE
               Up: 7,  Down: 2
```

Should work for you as well.

Did it stop transmitting or only stop serial output?
What does console Live traffic show?
Maybe you have poor (breadboard) contacts?
Does connecting (and enabling) a SSD1306 OLED display show uplinks continuing?

---

> Unfortunately, the memory is too small to use the new MCCI library.

Yes, for using MCCI LMIC with LMIC-node (or other larger applications) the memory on ATmega328 and ATmega32u4 is just too small. You _can_ run MCCI LMIC's included ttn-otaa.ino example on these MCU's however (also works on V3 stack). 

Don't forget to add setting clock error to 30000 ppm (1965) in ttn-otaa.ino or else it won't join.
See below example/extract for `setup()`:

```cpp
    // LMIC init
    os_init();
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    #define LMIC_CLOCK_ERROR_PPM 30000
    #if LMIC_CLOCK_ERROR_PPM > 4000
        // Allow clock error percentage to be > 0.4%
        #define LMIC_ENABLE_arbitrary_clock_error 1
    #endif    
    uint32_t clockError = (LMIC_CLOCK_ERROR_PPM / 100) * (MAX_CLOCK_ERROR / 100) / 100;
    LMIC_setClockError(clockError);
```

After compiliation PlatformIO shows the following memory usage (still not very much room but it works):

```text
RAM:   [========  ]  78.4% (used 1605 bytes from 2048 bytes)
Flash: [========= ]  93.5% (used 28716 bytes from 30720 bytes)
```

Output when running (uplinks are arriving in V3 console):

```text
Starting
Packet queued   
2650: EV_JOINING
377143: EV_TXSTART
698263: EV_JOINED
netid: 19
devaddr: 260BAF26
AppSKey: 13-C1-77-1B-81-F4-6D-5D-19-76-03-41-F5-D6-D8-5A
NwkSKey: CD-AF-72-BE-45-6D-48-DA-38-D2-18-17-76-01-D6-91
699360: EV_TXSTART
1081085: EV_TXCOMPLETE (includes waiting for RX windows)
2331632: EV_TXSTART
Packet queued      
2713362: EV_TXCOMPLETE (includes waiting for RX windows)
3963814: EV_TXSTART
Packet queued      
4345544: EV_TXCOMPLETE (includes waiting for RX windows)
5595995: EV_TXSTART
Packet queued      
5977725: EV_TXCOMPLETE (includes waiting for RX windows)
7228177: EV_TXSTART
Packet queued      
7609905: EV_TXCOMPLETE (includes waiting for RX windows)
```

---

> Was anybody successful to use an "Arduino-like" board on the V3 stack?

Yes, twice as described above. 🙂

al1fch, 2021-07-23:
Hi @sensorslot and @lnlp !!

This thread helped me to get a Pro Mini work on TTS V3 stack (tested OK with OTAA)
For the test I use the recent MCCI LMCIC V4 library and ttn-otaa.ino exemple from here 
Adding a DIO1 connection was necessary (DIO1  was unconnected before for  ABP mode, TinyLora Library, TTN V2, single chanel
Clock error set to to 30000 ppm (this thred pointed me to that point !!)

So Mega 328(pro mini) and TTS V3 are OK, I can confirm with your help !!

Best Regards

SensorsIot, 2021-07-25:
Thank you for both answers. I tried now again with the 328 chip as well as with TTGO boards and discovered strange things: Sometimes the payload is transferred ok with both boards (using this sketch, 328 with the old library, and the TTGO with the new library). First, I tried with OTAA. Often, I saw that the join request was accepted by TTNV3 but my node did not transmit the payload. I have to say: I want to use deep sleep and for the moment I often reset my board to simulate this effect. Unfortunately, I did not find a deep-sleep example that does not re-join every time. Such an example would be extremely important because most LoRa sensors use deep sleep.
So I moved to ABP. This used to work ok with V2. Unfortunately, I get a lot of times "Uplink not scheduled because TxRx pending". This, BTW, appears now also when I use OTAA. Very strange.

lnlp, 2021-07-26:
> using _this sketch_, 328 with the old library, and the TTGO with the new library

With _this sketch_ you mean **LMIC-node**?

> I tried now again with the 328 chip as well as with TTGO boards

Again, what 328 board exactly? 
Which TTGO boards?

Previous questions:

> What 'LoRa Radio Node board with an 328 chip' exactly did you use?

> Did you:
> 
> * make any changes to LMIC-node other than the configuration steps described in the Instructions chapter in README.md?
> * verify if pin mappings in the BSF and the actual hardware connections are correct and match?
> * check if the node actually hangs or only serial output stopped?
> * check Live data in the TTS V3 console for join requests/accepts and uplinks?
>   what does it show?

You seem not willing to provide details asked for. Without details, exact type of board, console data etc. it will be very difficult to try help determine what the possible cause of your problem is.
<br>

> I want to use deep sleep and for the moment I often reset my board to simulate this effect.  Unfortunately, I did not find a deep-sleep example that does not re-join every time. Such an example would be extremely important because most LoRa sensors use deep sleep.

Deep sleep behavior is very dependent on the MCU family and often also dependends on the rest of the hardware on the (development) board used. As LMIC-node is a releatively basic cross-platform application for showing how to use the LMIC library, adding support deep sleep for all kinds of platforms would be a huge project in its own. Also the (MCCI) LMIC library currently does not have some builtin generic support to save and restore LMIC state that could be used for (e.g. ESP32) deep sleep. Therefore deep sleep is not supported in LMIC-node. 

> I often reset my board to simulate this effect

I guess these simulations mainly target behavior of ESP32 chips which do not maintain their memory/state during deep sleep.

> Unfortunately, I did not find a deep-sleep example that does not re-join every time.

Possibly the _ttn-esp32_ library mentioned in below topic contains an example and can provide help for LMIC with ESP32 deep sleep: https://www.thethingsnetwork.org/forum/t/overview-of-lorawan-libraries-howto/24692
<br>

> So I moved to ABP. This used to work ok with V2.

Please note that ABP on V3 is not exactly the same as on V2. It is more strict and requires additional device settings to be specified when adding the device in the TTS console (or via CLI tools). In the console, part of these required parameters must be manually added; they are not automatically set by default.
For more information see topics in the [V2 to V3 Migration](https://www.thethingsnetwork.org/forum/c/v2-to-v3-upgrade/migrating-devices-to-v3/104) category and the [How to Migrate ABP Devices from V2 to V3 guide](https://www.thethingsnetwork.org/forum/t/how-to-migrate-abp-devices-from-v2-to-v3/43705).

<br>
For now (and without additional details) your problems appear not to be caused by an issue in LMIC-node.

al1fch, 2021-07-26:
`Unfortunately, I did not find a deep-sleep example that does not re-join every time. `
This  one seems  OK, tested ; No re-join, RTC Ram is used to save parameters
found 8mA during deep-sleep
[https://jackgruber.github.io/2020-04-13-ESP32-DeepSleep-and-LoraWAN-OTAA-join/]

closed 2021-07-27 by lnlp

SensorsIot, 2021-07-28:
You were right. I found the problem (it took quite a while and is the reason for my late answer): DIO1 was connected to a different pin (from an old project). Now OTAA works with the TTGO board.
Thank you for the lint to the files with deep sleep. I will try to add this knowledge to my project.
I did not migrate my devices, I created new ones (I only have a few) to avoid any migration problems.

lnlp, 2021-07-28:
Thanks for your feedback.

> I did not migrate my devices, I created new ones (I only have a few) to avoid any migration problems.

The link to the "How to migrate ABP Devices..." contains information that may still be useful to you.
It contains information that is currently difficult (or not yet) to find in the official documentation.
Especially the extra parameters that need to be explicitly specified in `Network layer | Advanced MAC settings` when adding a new ABP device (which is not done by default) e.g. explicitly adding `Factory preset frequencies`.

SensorsIot, 2021-07-29:
And now I got it also working on the pro mini 3.3v board. Thank you, lnlp: The sketch really needed your addition

#define LMIC_CLOCK_ERROR_PPM 30000
    #if LMIC_CLOCK_ERROR_PPM > 4000
        // Allow clock error percentage to be > 0.4%
        #define LMIC_ENABLE_arbitrary_clock_error 1
    #endif    
    uint32_t clockError = (LMIC_CLOCK_ERROR_PPM / 100) * (MAX_CLOCK_ERROR / 100) / 100;
    LMIC_setClockError(clockError);

Unfortunately, I do not know what it is for:-( But it worked also with my TTGO board. What is the reason it is not included in the example file?

I created an example that stores the LMIC variables to EEPROM (I switch the board completely off and it starts only once a day).
Your links helped a lot. Thank you again. Now I have the needed examples for my video. Next is the migration of my gateway to V3...

TD-er, 2021-07-29:
@Sensorslot
The clock error is to make sure the receive window isn't missed.
It requires quite an accurate timing to have the radio on on the right moment to receive the downlink messages.
So it is some trade-off between the lowest energy usage and making sure you don't miss downlink messages.

Maybe LMIC also uses it on a lower level to allow for a higher tolerance in tracing the chirp frequency hopping?
If so, then it might also affect the actual sensitivity.
[here](https://github.com/lnlp/LMIC-node#45-board-support-files) is some very short description of it.

lnlp, 2021-07-29:
> What is the reason it is not included in the example file?

If with "the example file" you mean `ttn-oota.ino` included with the MCCI LMIC library: 

1. That question would best be asked in the LMIC MCCI Github repository.

2. Setting ClockError is exceptional and should normally not be done. It loosens LMIC timing (makes timing less strict, less conforming to LoRaWAN specs) which normally has negative effects. It is needed for some slower 8-bit AVR MCUs but should not be used for anything else (unless practice has shown it is needed).

3. The examples included with the MCCI LMIC library are more primitive and less complete than LMIC-node _(see below excerpt from MCCI LMIC README.md)_. Other than LMIC-node they do not support a wide range of boards out of the box without having to make hardware dependent changes to the code.
_LMIC-node includes board-specific support for ClockError because it supports some boards that need it._

> Excerpt from MCCI LMIC library README.md:
> _The examples in this library (apart from the compliance sketch) are somewhat primitive. A very complete cross-platform Arduino application based on the LMIC has been published by ... (@lnlp) as LMIC-node_.

    
<br>

> I do not know what it is for ... But it worked also with my TTGO board.

DO NOT set ClockError for boards that do not need it (e.g. TTGO boards) as it will only make things worse, not better.

For all boards supported by LMIC-node you can check if ClockError is used (set) or not by checking the board's Board Support File (BSF). _The README.md states to check BSF files for important information about a board._
Much efforts and time went into writing and testing LMIC-node so that it is simple and effortless for others to use, out of the box. Determining if setting ClockError (and which value) for a board is needed was part of those efforts.

Excerpt from the `pro8mhzatmega328` BSF:

```cpp
// LMIC Clock Error
// This is only needed for slower 8-bit MCUs (e.g. 8MHz ATmega328 and ATmega32u4).
// Value is defined in parts per million (of MAX_CLOCK_ERROR).
// Value 30000 was determined empirically.
#ifndef LMIC_CLOCK_ERROR_PPM
    #define LMIC_CLOCK_ERROR_PPM 30000
#endif   
```

Excerpt from the `bsf_ttgo_lora32_v2` BSF:

```cpp
// LMIC Clock Error
// This is only needed for slower 8-bit MCUs (e.g. 8MHz ATmega328 and ATmega32u4).
// Value is defined in parts per million (of MAX_CLOCK_ERROR).
// #ifndef LMIC_CLOCK_ERROR_PPM
//     #define LMIC_CLOCK_ERROR_PPM 0
// #endif   
```

---

I've given some tips to make MCCI LMIC work with 8-bit AVR MCU's.
However, let's not forget that this repository is about LMIC-node and while LMIC-node uses the MCCI LMIC library, this repository is not the place to discuss the internals of the MCCI LMIC library and its included examples.

SensorsIot, 2021-07-29:
Thank you for your explanations. Now I understand better. I like your code and it runs well on ESPs. A lot of people still used 8 bit CPUs for their projects because the ESPs use quite some power. This is why I searched for a solution to use the newest LMIC library and V3. I learned that I have to recommend the OTAA example for this situation, which is ok with me.
I am glad you explained things which are not easily found and maybe it will help others to save time. Now, this thread is closed ;-)

lnlp, 2021-07-29:
> because the ESPs use quite some power

It's a pity that the ESP32 is often blamed for being power hungry, while this true when run at full speed and using Wifi/BT.
When lowering clock speed to 10 MHz and not use Wifi/BT it is [possible to get 8 mA power consumption](https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-3/18436/298) during operation (assumed you don't let it crunch numbers or mine bitcoins).

For deep sleep it should be possible to go as low as 10 to 20 uA. But unfortunately _most_ (Chinese) ESP32 development boards are not designed for efficient battery powered operation using deep sleep (but that you already know 🙂).

Maybe some case for one of your videos?: A bare ESP32 module with 4 MB flash without PSRAM, running at 10 MHz, not Wifi/BT, with SX1276/RFM95 LoRa module and MCCI LMIC library.

SensorsIot, 2021-07-29:
I agree with your analysis that the board often is the problem and not the chip. Maybe it would be an idea to do a test for the best LoRa board. Contenders would then also be the Pi Pico, an 328p, and maybe an STM Blue or Blackpill and an ESP32.
I created a Mailbox notifier that is completely switched off and started with a magnetic switch. This is why I needed a sketch that does not join all the time... And I was able to use a not-so-good board. 
I still think you should include something for deep sleep/restart and OTAA because I do not know of many nodes running all the time.

TD-er, 2021-07-29:
Not sure if the MCCI LMIC library allows access to the OTAA keys etc.
If it does and allows to restore them, then it would be a great idea to support it in this library (I also would like it for ESPEasy ;) )
Currrently I'm only aware of the RN2483 which allows to save/restore OTAA activation.

lnlp, 2021-07-29:
> I still think you should include something for deep sleep/restart and OTAA because I do not know of many nodes running all the time.

I think many people would like that.
However, as said before doing that cross-platform for all supported MCU families would be a huge project on its own.
Also MCCI LMIC currently does not provide any generic hooks for saving and restoring LMIC state. This will make a custom implementation more brittle and possibly none-future-proof (and hence maintenance intensive).

My goal of finishing and publishing LMIC-node was to make using LMIC simple and accessible to a wide audience _(based on my own experiences from the past few years)_ without running into invisible walls and having to go through steep learning curves first. Based on the feedback LMIC-node succeeded in that goal quite well.

SensorsIot, 2021-07-29:
For the moment I only saved the LMIC structure in EEPROM and reload it after boot (another project saves it in RTC memory for deep sleep). For the moment this works. But I only have a few hours of experience. I also saw a parameter that joins automatically from time to time. I plan to investigate this matter a little more.

lnlp, 2021-07-29:
> Not sure if the MCCI LMIC library allows access to the OTAA keys etc.

It already reads and stores all OTAA and ABP related LoRaWAN keys so that should not be an issue.

> it would be a great idea to support it in this library

This functionality [is on the MCCI LMIC wish list](https://forum.mcci.io/t/lmic-development-plans/19/2) to be implemented but no specific plans have been made available yet.