lnlp/LMIC-node #18: Example code not working on Heltec LoRa 32 v2

xatru, 2021-08-04:
Greetings,

and first of all thanks for that great project and the detailed description on the front page. I just startet to work with LoRa a few days ago and I still have many questions, but the documentation is quite good. Unfortunately, the example code does not work for me. The compilation and upload works fine, but the only thing I can see on the TTB console is a "Accept join-request", after around one minute. 

The serial console of LoRa 32 (with debugging flag set to 2) shows the following:

`RXMODE_RSSI
000000029396:  Event: EV_JOINING
31440: engineUpdate, opmode=0x4 

000000031474:  doWork job started
35075: engineUpdate, opmode=0x4
000000035115:  Event: EV_TXSTART
37234: TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5, IH=0
352849: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 2
353479: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 354007 rxtime: 353474 entry-rxtime: 533 now-entry: 2 rxtime-txend: 312375
415349: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 1
415978: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 430330 rxtime: 415974 entry-rxtime: 14356 now-entry: 2 rxtime-txend: 374875
430345: processRx2Jacc txrxFlags 0x2 --> 00
000000430388:  Event: EV_JOIN_TXCOMPLETE
432883: engineUpdate, opmode=0x4
<snip>
`
What did I do?
 I installed Visual Studio code and PlatformIO. Then i cloned the GitHub repo and opened it as a project. I selected the board "heltec_wifi_lora_32_v2" in the `platformio.ini` and set the correct LoRaWAN keys in `lorawan-keys.h`, which is a copy of the examples key file.  I also added the uplink formatter to the TTN console. Compile and upload worked fine (dependencies installed on the first build). The display of the LoRa 32 shows the following:

`
LMIC-node
wifi-lora-32-v2
Interval: 60s

Time:21040164
TE_JOIN_TXCOMPLE
`

The event details for the join-request event are:
`{
  "name": "js.join.accept",
  "time": "2021-08-04T11:24:32.032208112Z",
  "identifiers": [
    {
      "device_ids": {
        "device_id": "eui-70b<blanked>",
        "application_ids": {
          "application_id": "<blanked>"
        }
      }
    },
    {
      "device_ids": {
        "device_id": "eui-70b3<blanked>",
        "application_ids": {
          "application_id": "<blanked>"
        },
        "dev_eui": "<blanked>",
        "join_eui": "0000000000000000",
        "dev_addr": "<blanked>"
      }
    }
  ],
  "correlation_ids": [
    "rpc:/ttn.lorawan.v3.NsJs/HandleJoin:01FC8EGTCTR1VE2CMQDQQC6MTH"
  ],
  "origin": "ip-<blanked>.eu-west-1.compute.internal",
  "context": {
    "tenant-id": "CgN0dG4="
  },
  "visibility": {
    "rights": [
      "RIGHT_APPLICATION_TRAFFIC_READ",
      "RIGHT_APPLICATION_TRAFFIC_READ"
    ]
  },
  "unique_id": "<blanked>"
}`

Any idea what's wrong here?

Thanks!

lnlp, 2021-08-04:
Hi,

The log from the serial monitor shows the following:

Eevent EV_JOINING.
This means that the node sent a _join request_.

Later you see event EV_JOIN_TXCOMPLETE.
This means that the node did not receive (not recognize) a _join accept_ response in time.
_When the join succeeds you should see event EV_JOINED in the output._

A possible cause for joins to fail is that downlinks are not received/recognized by the node.
A common cause for failing downlinks is when DIO1 from the LoRa radio is not connected  to the correct GPIO port or is mapped incorrectly in the software. On the Heltec Wifi LoRa 32 (V2) boards DIO0, DIO1 and DIO2 are allready wired on the PCB (no manual wiring required).

Is your board really a 'Heltec Wifi LoRa 32 V2' or maybe the older V1 which has boardid `heltec_wifi_lora_32` (or even a completely different board)?
On the V2 board DIO1 and DIO2 are connected to different ESP32 GPIO ports than on the V1 board.

Using an incorrect boardid will cause failure. It may result in LMIC-node not working at all or just partially (e.g. uplinks working but downlinks fail due to incorrectly mapped DIO1). The boardid determines which BSF and what settings are used (including the pin mappings).
<br>

FYI:

The first part of your serial output is missing (it should start with "LMIC-node" followed by some header information).
It is usually preferred to show the serial log from the start.

The best way to capture all serial output is do the following (as described in the Tips chapter), in VSCode:

1. After having the board connected, start the VSCode serial monitor.
This will open the serial monitor window.

2. Upload the sketch from within VSCode. This will show the upload window, but after the upload it will automatically switch back to the serial monitor window.

This way you will not miss the first output sent to the serial monitor.

xatru, 2021-08-04:
Hi lnlp,

and thanks for the quick response. At the backside of my board `WiFi_LoRa_32_V2` is printed and at the other side also Heltec, so I guess it's really the correct board. The case were the board were delivered in looks like on this [picture](https://hardverapro.hu/dl/uad/2020-04/4639184/img_20200409_174617.jpg).

I also deleted the existing device in TTN and created a new one. Of course I updated the key-files, compiled again, started the console and uploaded the program. Here are the new outputs after approximately half an hour running the program. 

Debug output:
```
LMIC-node

Device-id:     wifi-lora-32-v2
LMIC library:  MCCI
Activation:    OTAA
Interval:      60 seconds     

000000029360:  Event: EV_JOINING 

000000031403:  doWork job started
000000310833:  Event: EV_TXSTART
000000706088:  Event: EV_JOIN_TXCOMPLETE

000003781403:  doWork job started
000004446640:  Event: EV_TXSTART
000004841932:  Event: EV_JOIN_TXCOMPLETE

000007531403:  doWork job started
000008919150:  Event: EV_TXSTART
000009314442:  Event: EV_JOIN_TXCOMPLETE

000011281403:  doWork job started
000013585061:  Event: EV_TXSTART
000013983614:  Event: EV_JOIN_TXCOMPLETE

000015031403:  doWork job started

000018781403:  doWork job started
000021750815:  Event: EV_TXSTART
000022149367:  Event: EV_JOIN_TXCOMPLETE

000022531403:  doWork job started

000026281403:  doWork job started
000029609455:  Event: EV_TXSTART
000030008008:  Event: EV_JOIN_TXCOMPLETE

000030031403:  doWork job started

000033781403:  doWork job started
000036910466:  Event: EV_TXSTART
000037314809:  Event: EV_JOIN_TXCOMPLETE

000037531403:  doWork job started

000041281403:  doWork job started

000045031403:  doWork job started

000048781403:  doWork job started
000050682805:  Event: EV_TXSTART
000051087151:  Event: EV_JOIN_TXCOMPLETE

000052531403:  doWork job started

000056281403:  doWork job started

000060031403:  doWork job started

000063781403:  doWork job started
000065019602:  Event: EV_TXSTART
000065423948:  Event: EV_JOIN_TXCOMPLETE

000067531403:  doWork job started

000071281403:  doWork job started

000075031403:  doWork job started

000078781403:  doWork job started
000081516351:  Event: EV_TXSTART
000081931001:  Event: EV_JOIN_TXCOMPLETE

000082531403:  doWork job started

000086281403:  doWork job started

000090031403:  doWork job started

000093781403:  doWork job started

000097531403:  doWork job started

000101281403:  doWork job started

000105031403:  doWork job started
000107776836:  Event: EV_TXSTART
000108191530:  Event: EV_JOIN_TXCOMPLETE

000108781403:  doWork job started

000112531403:  doWork job started

000116281403:  doWork job started

000120031403:  doWork job started

000123781403:  doWork job started

000127531403:  doWork job started

000131281403:  doWork job started
000132690273:  Event: EV_TXSTART
000133104969:  Event: EV_JOIN_TXCOMPLETE

000135031403:  doWork job started

000138781403:  doWork job started

000142531403:  doWork job started

000146281403:  doWork job started
```

TTN console data:
[image]

lnlp, 2021-08-04:
You have not yet answered my questions about gateway and distance between node and gateway. Please do.

> At the backside of my board `WiFi_LoRa_32_V2` is printed and at the other side also Heltec, so I guess it's really the correct board. The case were the board were delivered in looks like on this [picture](https://hardverapro.hu/dl/uad/2020-04/4639184/img_20200409_174617.jpg).

So it looks like it is a real Heltec Wifi LoRa 32 V2. I am not familiar with that packaging but things change and improve over time (my V2 packaging was different but is probably 2+ years old).

My V2 also has "WiFi LoRa 32 V2" and "Heltec" on the bottom side of the PCB and "V2" on top (see picture below):

[image]

Does your board look similar/identical?
<br>

Your serial log clearly shows that the OTAA join is failing and retrying but does not succeed.
The console shows that the join is accepted by the backend. The _join accept_ downlink is not received by the node however.
<br>

You could try to use ABP activation instead of OTAA.
This will allow you to tests if you can send uplinks succesfully (ABP does not require downlinks to function for that).
If you have uplinks working over ABP next step is to try a downlink (can be done from console, my docs also give some hints).
If then the downlinks will fail the issue will be clearly identified (but not yet what causes it).

If you get stuck with any of these steps please ask for help on The Things Network forum. There is a wealth of information available and there are experienced users on the forum that may help you solve your problem.

For ABP you will have to create a new device in the console which for ABP can be a bit tricky because for V3 some advanced device parameters will hve to be manually added in the console. See issue #17 for links to some V3 and ABP related topics.

xatru, 2021-08-04:
```
Are you using a LoRaWAN complaint 8-channel gateway?
Is there sufficient distance between node and gateway (3m at very minimum, you may try more, e.g. 10+m).
```

I am not sure about the gateway manufacturer, as it is hosted by one of my neighbors. However, he has already several LoRa applications running, most of them realized with the [Octopus](https://www.tindie.com/products/FabLab/iot-octopus-badge-for-iot-evaluation/)baord. The gateway is around 800m away. The area here around me is still in good coverage. 

> Does your board look similar/identical?
The board looks absolute identical. 

I will try the ABP activation this afternoon. Let's see if that works. Again, thank for you help.

lnlp, 2021-08-04:
It is essential that the gateway is a normal 8-channel LoRaWAN compliant gateway.
So called 'single channel packet forwarders' (SCPF) that some people used (in the past) are not LoRaWAN compliant and can also cause joins to fail (and will not work with LMIC-node).

> However, he has already several LoRa applications running, most of them realized with the [Octopus](https://www.tindie.com/products/FabLab/iot-octopus-badge-for-iot-evaluation/)baord.

LoRa or LoRaWAN applications? That is an important difference.

The linked Octopus board itself does not have LoRa support and will require an external LoRa radio module for LoRa(WAN). 
What LoRa radio module and what software are used are not clear from the article and doesn't say anything about the gateway.

lnlp, 2021-08-13:
No response, therefore closing the isssue.

closed 2021-08-13 by lnlp