lnlp/LMIC-node #20: MIC mismatch with ttgo lora

Bighoneypot, 2021-10-18:
Hi, with this firmware I have always error in TTN:
Join-request to cluster-local join server failed MIC mismatch

Do you have a idea?

Can you hel me? Thank you!

on terminal:

Device-id:     ttgo-lora32-v1
LMIC library:  MCCI
Activation:    OTAA
Interval:      60 seconds    

000000023153:  Event: EV_JOINING

000000023943:  doWork job started
000000161248:  Event: EV_TXSTART
000000555186:  Event: EV_JOIN_TXCOMPLETE

lnlp, 2021-11-03:
@Bighoneypot
Please respond.
Did above help to solve your issue?

Bighoneypot, 2021-11-03:
@lnlp 
Gday,
yes I solve my issue with remove entire application and after I create new application the device work fine.

lnlp, 2021-11-04:
Thanks. Good to hear your issue has been solved.

closed 2021-11-04 by lnlp