lnlp/LMIC-node #22: Not a bug

mrbc42, 2021-11-09:
Thank you for your hard work on this, it is literally the only code out there from probably 10 libraries I have tried that works first time on the T-Beamv1.0

Is there any plans to add GPS functionality to your code so we can use it on TTN-Mapper?  Unfortunately i'm not a coder and have no idea how to do this.  Do you know if I can drop in a few files from the below and get it working?

https://github.com/kizniche/ttgo-tbeam-ttn-tracker

thanks again,

Matt

lnlp, 2021-11-10:
Thanks for your compliments!

> Is there any plans to add GPS functionality to your code so we can use it on TTN-Mapper?

No plans currently to add this to LMIC-node, because it would be specific to only a few of the supported boards while making the code more complex. My goal for LMIC-node was to serve as a basic but functionally complete cross-platform example that runs out of the box on many popular boards, to make building LoRaWAN nodes with LMIC and Arduino easier and more fun.

However: @roelwolf already added the GPS functionality that you are looking for and has tested it on TTGO T-Beam V1.0. 🙂

The README.md from his repository unfortunately has still not been updated. So from the README.md it does not become clear what functionality was added or changed, which hardware is supported and how to use it. You will have to look at the code for that. Some additional information can be found in the following [post on the TTN forum](https://www.thethingsnetwork.org/forum/t/lmic-node-one-example-to-rule-them-all/46964/26)

For the repository see: https://github.com/roelwolf/LMIC-node-gps-tracker

mrbc42, 2021-11-11:
> However: @roelwolf already added the GPS functionality that you are looking for and has tested it on TTGO T-Beam V1.0. slightly_smiling_face

This is really great, I got it working as easy as your code.  Now I can use TTN Mapper.  For others looking to do the same thing, @roelwolf doesn't provide the payload formatter for the GPS data.  This is easily fixed by copying the formatter from the link in my original question which works fine.

appreciate all the help

closed 2021-11-11 by mrbc42

roelwolf, 2021-11-11:
@mrbc42 A payload formatter is included in my repository [(**link**)](https://github.com/roelwolf/LMIC-node-gps-tracker/blob/main/payload-formatters/decodeUplink-v3.js)
Do not use the formatter you provided as this formatter also tries to convert downlink messages on port 10 to a gps position, resulting in positions all over the world.

mrbc42, 2021-11-11:
@roelwolf  many thanks, for some reason I missed the formatter in your repo.  I have corrected this in my system (working fine) and will edit my original statement.  Thanks for the great code, it solved my problem.