lnlp/LMIC-node #23: strange behaviour, application crashing sometimes...

askmurphy, 2021-11-10:
My application (T-Beam LoRa v1.1) is crashing sometimes, not sure why:

000032937399:  Packet queued
000032937523:  Event: EV_TXSTART
FAILURE
.pio\libdeps\ttgo_t_beam_v1\MCCI LoRaWAN LMIC library\src\lmic\radio.c:1065

On the board, I have connected 3 push-buttons, connected to pin 2, 4 and 32.
Pinmode:  INPUT_PULLUP and to be sure, also attached a 10K resistor between datapin and 3.3Volt.

Will add more details here asap.

The crash appears sometimes after pressing the pin32-button.

lnlp, 2021-11-10:
GPIO32 is connected to DIO2 of the LoRa radio and used by LMIC. GPIO32 shall not be used for other purposes. 
This may explain your crashes.

Check file `bsf_ttgo_t_beam_v1.h` for a description of the pin mappings.

For pullups better use either internal or external pullups but not both.

askmurphy, 2021-11-10:
Thanks!
Changed the button pin to gpio13, internal pullup option removed, problem solved ;-)

closed 2021-11-10 by askmurphy