lnlp/LMIC-node #24: Uplink not scheduled because TxRx pending

alexinthis, 2021-11-13:
Hi,

I am trying to use this code on the Helium network and have made the lease amount of modifications possible to get this to work. 

I am able to receive the first transmit and after that it just says: "UL not scheduled" on the OLED or "Uplink not scheduled because TxRx pending" in the serial output.

My hardware is a ttgo_lora32_v2, and I am in Australia so have "-D CFG_au915=1" uncommented. My keys must be ok as I am receiving Join and Accept messages plus the first data message. 

If I had to guess the first send never completes correctly. 
I would normally use TTN but there is no gateway near me and someone has one setup on Helium. I don't really have the budget to purchase a gateway for either. 

Here is an output of my serial monitor: 

> 
> 000000027651:  Event: EV_JOINING
> 000000029610:  Event: EV_TXSTART
> 
> 000000031588:  doWork job started
> 000000372546:  Event: EV_JOINED
>                Network Id: xxx
>                Device Address: xxx
>                Application Session Key: xxxxxxxx
>                Network Session Key:     xxxxxxxx
> 
> 000003781588:  doWork job started
> 000003784738:  Input data collected
>                COUNTER value: 1
> 000003785570:  Packet queued
> 000003787704:  Event: EV_TXSTART
> 
> 000007531588:  doWork job started
> 000007534738:  Input data collected
>                COUNTER value: 2
> 000007534738:  Uplink not scheduled because TxRx pending
> 
> 000011281588:  doWork job started
> 000011284738:  Input data collected
>                COUNTER value: 3
> 000011284738:  Uplink not scheduled because TxRx pending

alexinthis, 2021-11-14:
For anyone reading this who has my same issue you do in fact need to solder the DIO1 pin to GPIO33. It's working after making this change.

This link has the pin mapping you'll need
https://github.com/LilyGO/TTGO-LORA32/issues/3

closed 2021-11-14 by alexinthis

lnlp, 2021-11-14:
@alexinthis 

What you describe is already **perfectly documented** in LMIC-node's [README.md](https://github.com/lnlp/LMIC-node#21-lora-development-boards) and in the board's BSF file [`bsf_ttgo_lora32_v2.h`](https://github.com/lnlp/LMIC-node/blob/38b1804062cd272a31f50de4ba474bfbfcae8155/src/boards/bsf_ttgo_lora32_v2.h#L17)!

> This link has the pin mapping you'll need
> [LilyGO/TTGO-LORA32#3](https://github.com/LilyGO/TTGO-LORA32/issues/3)

No, that is not where anyone should look for pin mappings because everything is already contained and documented in LMIC-node.

A _newer_ version of the pin-mapping document for the TTGO LoRa32 V2.0 can be found in my [pinout-diagrams](https://github.com/lnlp/pinout-diagrams) repository, which is also (clearly) mentioned in LMIC-node's README.md.

So next time better RTFM first!! 😉