lnlp/LMIC-node #27: ttgo-lora32-v1

kskenyon, 2022-01-26:
My board does have an onboard blue led at pin 2.
I had to add rst (pin 16) to the display call to get it to work.

Nice app overall.

K

lnlp, 2022-01-31:
Thanks.

Do you know the exact version of your TTGO LoRa32?
Can you provide a picuture of the top and bottom sides?

I don't have any of the TTGO LoRa32 v1.x boards myself and therefore have to rely on LilyGo's documentation.
Unfortunately their versioning is unclear (sucks) and documentation is poor and unreliable.

The TTGO LoRa32 v1.3 schematic diagram does not show a LED nor seems the display's reset to be connected to a GPIO pin.
Is it possible that you have an older version than v1.3? (With the v1.3 board the vesion in printed on PCB bottom side).

If you have an older version I could add an additional board definition to LMIC-node for older (pre-v1.3) TTGO LoRa 32 v1.x boards and add the proper display reset and LED pin definitions.

kskenyon, 2022-01-31:
> Can you provide a picture of the top and bottom sides?

Thanks for getting back to me.  It doesn’t sound like I have a v1.3 version.  The blue LED is on.  For some reason it works the opposite of the transmit flag.  I tried some different code in LMIC-node to get it to work properly but I wasn’t successful. Weird.

Mine is 2 or 3 years old.  The display is pooping out as you can see.  No indication of version. Here is the pin-out diagram that matches it.

kskenyon, 2022-02-14:
> There's nothing to see. It looks like you forgot to attach the images.

They were attached to the return email. 

Fatfingered from my iPhone

kskenyon, 2022-02-14:
Maybe too big for GitHub. 

Fatfingered from my iPhone

lnlp, 2022-02-15:
You can add them in the regular way via the GitHub site (or possibly via the GitHub app).
If too big please resize them.

lnlp, 2022-02-24:
Thanks for adding the pictures.

Your board looks like the predecessor of TTGO LoRa32 v1.3 as shown here: [Big ESP32 topic](https://www.thethingsnetwork.org/forum#ttgo-boards-28)

Let's call that version v1.0, because version information about pre-v1.3 boards is not available from LilyGO.
I can add a separate board definition for v1.0 version (in addition to the v1.3 version) to LMIC-node.

> It doesn’t sound like I have a v1.3 version. The blue LED is on. For some reason it works the opposite of the transmit flag.

Can you test if the LED on GPIO2 on your board is active-low or active-high?

The following library and example can help with that: [EasyLed active-level tester](https://github.com/lnlp/EasyLed/tree/main/examples/led-activelevel-tester)

In

```cpp
EasyLed led(LED_BUILTIN, EasyLed::ActiveLevel::Low, EasyLed::State::Off);  //Use this for an active-low LED
// EasyLed led(LED_BUILTIN, EasyLed::ActiveLevel::High, EasyLed::State::Off);  //Use this for an active-high LED
```

replace `LED_BUILTIN` with `2`.

kskenyon, 2022-02-24:
I’m traveling and won’t be able to do that for a while. I tried it both ways in your code previously and it still worked backward. Didn’t troubleshoot further.

lnlp, 2022-02-24:
I would appreciate if you can check it when possible again.
Thanks.

lnlp, 2022-02-27:
@kskenyon 
I have created an issue to add support for the TTGO LoRa32 V1.0, see #31.

For this I would like to know if the LED on GPIO2 is active-low or active-high.
Hope you can provide that information when possible for you again.

JohnMacrae, 2022-04-23:
I have this board. The LED is active-high.  The main problem is getting the display to work. V frustrating - I'd appreciate any pointers you have - prepared to get into it but stumped at the moment...

JohnMacrae, 2022-04-23:
Rereading the comments above, con confirm that adding RST pin 16 allows display to operate.  Will clone and push tomorrow.

lnlp, 2022-04-30:
@JohnMacrae Thanks for your feedback!

I will add a _separate_ board definition for TTGO LoRa 32 V1.0 later when I have time.

closed 2022-04-30 by lnlp