lnlp/LMIC-node #28: Question: How to properly include deepsleep functionality instead of timed rescheduling?

drfritz142, 2022-01-31:
Dear Leonel,

**thanks first and foremost for this very comprehensive software**. I could get it running on a bluepill/rfm95 within a few minutes. Just wonderful.

Now I would like to include a deepsleep function. Most of the time my sensor will be in deepsleep to save battery. Previously, I have just "hacked" that deepsleep call into the code right at the end of the EV_TXCOMPLETE event. But this seems nothing but brutal in your code. I bet there could be a place where it would be properly called within a wrapper function perhaps which every board then could specify out by itself. But where would that call be and how would one properly stop the timed rescheduling? Essentially, this should be replaced: instead of rescheduling after x seconds, the MC should sleep and when waking up, start the next roundtrip.

Could you be so kind as to let me know where I can implement this properly?
Thanks,
Andreas

drfritz142, 2022-01-31:
Of course, this is not an issue but a question. But somehow I didn't find a way to select this ...

lnlp, 2022-02-14:
Hi,

Thanks for your feedback.

Deep sleep support is a much wanted feature and very relevant for battery powered LoRaWAN devices.

Unfortunately LMIC-node currently does not include any deep sleep support.

The first (and main) reason is that the (MCCI) LMIC library still does not provide any support / hooks for deep sleep.
Support for functions like e.g. `saveLmicState()` and `restoreLmicState()` have yet to be added to the LMIC library (user will need to provide the actual implementation).
Another issue is that the timers that LMIC depends on, need to updated after deep sleep. This is essential because LMIC keeps track/manages which channels it is when allowed to transmit on. The LMIC library currently does not provide support for this.

A second reason is that deep sleep behaves differently on different MCU platforms (e.g. ESP32 actually resets when waking up from deep sleep while other platform preserve all state). Without any generic support from the LMIC library this would be quite complex (and much work) to implement for all MCU platforms that LMIC-node currently supports.

drfritz142, 2022-02-14:
Hi Leonel,

 

thanks for your reply. I think it's an understatement, that deepsleep is very relevant - I believe it's essential! 

 

I found a solution which was realized for an ESP32 (which as you wrote forgets EVERYTHING during deepsleep - more like a coma really...):

 

https://jackgruber.github.io/2020-04-13-ESP32-DeepSleep-and-LoraWAN-OTAA-join/

 

I adopted it for my STM32, for the time being without an EEPROM to persist and restore the LMIC state, I just essentially pause the program at that point with an STM32 deepsleep function. After wakeup, STM32 still remembers the variables. It seems to work... Last two years we had many devices in the APB mode with this approach, essentially fire & forget. Now we moved to MCCI, TTN V3, and OTAA. 

 

I am not actively taking care of the timing thing you mentioned, I just let it go...... I wonder whether this will get me into trouble...

 

the STM32 deepsleep routine is woken up by a timer (so _one_ clock must still be running on the chip) but functions like millis() or so do not get advanced during the deepsleep. 

 

Thanks for your insights. Would you know when something is happening w/r to deepsleep on the MCCI front? Please let me know.

 

Best wishes...

 

 

 
 

Gesendet: Montag, 14. Februar 2022 um 17:11 Uhr

closed 2022-02-15 by lnlp