lnlp/LMIC-node #29: What are opmodes? Are they listed somewhere?

tipo1000A, 2022-02-24:
As the title says LMIC-node prints opmodes like:
opmode=0x4
opmode=0x808
opmode=0x900
opmode=0x908
. . ,
What they mean?

Thanks,
Tipo

lnlp, 2022-02-24:
That is functionality of the LMIC library.

You can find definition of the opmodes with a brief description in [lmic.h](https://github.com/mcci-catena/arduino-lmic/blob/8d378ea410887d3fb08ea2c9acce95cc4c047788/src/lmic/lmic.h#L256).

tipo1000A, 2022-02-25:
I can find opmodes 0x4 and 0x808 in lmic.h.

However, opmodes 0x900 and 0x908 are not in that list. How should I understand those modes?

lnlp, 2022-02-27:
> opmodes 0x900 and 0x908 are not in that list. How should I understand those modes?

> Opmodes are MAC operation modes. They are functionality of the LMIC library.

The best place for asking that question is the [MCCI LMIC library repository](https://github.com/mcci-catena/arduino-lmic).

closed 2022-03-06 by lnlp