lnlp/LMIC-node #32: maybe a small mistake? shouldn't it be #ifndef MCCI_LMIC   instead of #ifndef LMIC_MCCI

drfritz142, 2022-03-02:
could it be? it seems to do no harm if it is wrong at least i didn't encounter any problems either way... 

https://github.com/lnlp/LMIC-node/blob/4871fbe8ad507a05e7ef390a40dddabd97fc9246/src/LMIC-node.h#L151

lnlp, 2022-03-06:
Thanks, good catch! 

This is a typo/bug and should be `MCCI_LMIC` instead.

I did not notice it before because it seems to cause no visible negative effect (e.g. 'is already defined' type errors/warnings when using MCCI LMIC).

LMIC_MCCI is never defined in LMIC-node, therefore the code in the #ifndef block was always included.
The code is only needed for Classic LMIC. For MCCI LMIC it is already included in the library.

lnlp, 2022-03-06:
[Fixed](https://github.com/lnlp/LMIC-node/commit/fabdde20276baedb91526cf8943e7ad3f076b69d).

closed 2022-03-06 by lnlp