lnlp/LMIC-node #33: Persist OTAA session across reboots

giannoug, 2022-03-03:
First of all thanks for the project, it is a really good starting point for those new to LoRaWAN :)

My current issue is that my device can't hear for join messages after booting. I'll have to move the device outside first and then move it to its normal position to start transmitting messages. I've used ABP while testing with The Things Network but now that I switched to Helium (which only supports OTAA) this a huge issue.

I've read that I can persist the "network session key" and "application session key" on RTC memory on my ESP32 (I believe this would be an issue with all the board this project supports) but I'm a bit lost on whether this is correct or a simpler solution exists. Any thoughts?

lnlp, 2022-03-06:
Thanks.

Saving and restoring session keys is currently not supported / not included in LMIC-node.
The goal of LMIC-node is to serve as a basic example that works out of the box cross-platform.
Persisting state in non-volatile storage to survive power cycles, reboots or deep sleep is not supported by LMIC-node because this is microcontroller platform and hardware dependent and the LMIC library does currently not provide any generic support (functions/hooks for storing and restoring LMIC state) to help implement this.

> My current issue is that my device can't hear for join messages after booting. I'll have to move the device outside first and then move it to its normal position to start transmitting messages.

Looks like you have an RF connection issue to solve here as the device shall be able to both transmit and receive.
If you are unable to receive downlink messages, this does not only impact join accepts but also means that your device cannot receive essential MAC commands from the network server. For LoRaWAN, even for ABP the device still needs to be able to receive downlink messages. ABP is not a synonym for 'transmit-only' (although with TTN V2 many devices were implemented as such transmit-only).

giannoug, 2022-03-06:
>Looks like you have an RF connection issue to solve here as the device shall be able to both transmit and receive.
If you are unable to receive downlink messages, this does not only impact join accepts but also means that your device cannot receive essential MAC commands from the network server. For LoRaWAN, even for ABP the device still needs to be able to receive downlink messages. ABP is not a synonym for 'transmit-only' (although with TTN V2 many devices were implemented as such (transmit-only)).

Thanks for the clarification. I'm able to receive downlink messages but not in the place I'd like to install the device. Maybe I'll need to fix that issue with a better antenna(?) first and then look into saving and restoring session keys.

>Maybe the following fork of LMIC-node, for an ESP32 device used with Helium can provide you some help: [LMIC-node_Sensor-for-Helium-Network](https://github.com/Chiumanfu/LMIC-node_Sensor-for-Helium-Network)

I'll surely have a look for ideas 👀 Thanks!

lnlp, 2022-03-07:
> Maybe I'll need to fix that issue with a better antenna(?) first

I would try to fix that issue first. A different antenna could possibly. Success.

closed 2022-03-07 by lnlp