lnlp/LMIC-node #35: TTGO T-Beam V1.1 - analogRead - Uplink not scheduled because TxRx pending

JohanScheepers, 2022-03-12:
I have add the following (see below). It sends one uplink and then "Uplink not scheduled because TxRx pending", not sending any more uplinks (see serial print)

#define loadV 33

And then in --- if (LMIC.devaddr != 0) ---

            //Read Voltages

            int16_t int16_loadV;
            int16_loadV = 0;

            float Vload = analogRead(loadV);
            int16_loadV = Vload;
            Serial.print("int16_loadV: " );
            Serial.println(int16_loadV);

Serial print:
```
000000772779:  Event: EV_JOINED
               Network Id: 19
               Device Address: 2xxxxxx4
               Application Session Key: A5-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-CC
               Network Session Key:     7A-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-xx-1F

000000773629:  doWork job started
int16_loadV: 22
000000773829:  Input data collected
               downLink: 0
000000774275:  Packet queued
000000774492:  Event: EV_TXSTART

000002648629:  doWork job started
int16_loadV: 4095
000002648639:  Input data collected
               downLink: 0
000002648639:  Uplink not scheduled because TxRx pending
```

lnlp, 2022-03-13:
It's not clear to me.
Is this a question, a proposal for a fix or something else?

JohanScheepers, 2022-03-14:
Hi,

I can not do a analogRead, so it's is a fix that needs to be done.

As soon as you use analogRead, it stops transmitting any date on the LoRaWan side.

Thank you

lnlp, 2022-03-14:
> I can not do a analogRead

Sounds like a problem you have.

> so it's is a fix that needs to be done.

Still unclear. Is the fix in your above example? If so that is not clear at all.

> As soon as you use analogRead, it stops transmitting any date on the LoRaWan side.

Apparently there is some conflict between the use of analogRead, the LMIC library and your board.

Your post lacks essential details (e.g. your board, more detailed code).
This also does not sound like a LMIC-node specific issue so I cannot help you with that.
I suggest to use the TTN forum for more help instead.

In case you have a specific fix, I might be able to add it to LMIC-node.

JohanScheepers, 2022-03-14:
As soon as I add a analogRead to LIMC-Node, the issue occurs where it stops sending uplinks, tried it on 3 different boards (TTGO T-Bean V1.1) and the problem resists.

I then took a Adafruit Feather, this preforms the analogRead.

So it seems the problem might be the ESP32.

JohanScheepers, 2022-03-14:
> Your post lacks essential details (e.g. your board

TTGO T-Beam v1.0, v1.1

> detailed code

Add a analogRead in the section where 

> // Collect input data.
        // For simplicity LMIC-node uses a counter to simulate a sensor. 
        // The counter is increased automatically by getCounterValue()
        // and can be reset with a 'reset counter' command downlink message.

float Vload = analogRead(loadV);

lnlp, 2022-03-14:
I overlooked the board in the title, read it at first but then couldn't find it in the text.🤦‍♂️

Good you were able to also test it with a different board. 
It may be caused by several reasons but I have no clue. It seems to cause a timing issue that effects operation/timing of LMIC. This may be a general ESP32 related issue but could also be TTGO T-Beam specific, possibly related to its power management chip (configuration).

JohanScheepers, 2022-03-15:
I were making my references to their, LilyGO, board layout, they don't mention on the board layout the LoRa connections. Although I am full aware of the pin mapping in bsf_ttgo_t_beam_v1.h file.

Using analogRead on pin 2 and it works 100%.

PS. I do own a shotgun, I thing both of us used it ;-)

lnlp, 2022-03-15:
I'm glad I was able to help fix your issue.

As expected LMIC-node was not the culprit for your issue.
_A lot of preparation went into making LMIC-node work out of the box with its supported boards.
It helps prevent others having to go through a steep learning curve first and step into the many (at first sight invisible) possible pitfalls, of which poor product documentation from the manufacturer is only one.

closed 2022-03-15 by lnlp