lnlp/LMIC-node #36: Definition: _LMIC_CONFIG_PRECONDITIONS_H_

CK1-CK, 2022-03-13:
Hi

where is the definition to "_LMIC_CONFIG_PRECONDITIONS_H_"? 
I don't  know where to define the _LMIC_CONFIG_PRECONDITIONS_H_

The project compiles everytime with the deprecated lmic lib. Unfortunately the deprecated lib doesn't work with TTN Stack v3.

https://github.com/lnlp/LMIC-node/blob/fabdde20276baedb91526cf8943e7ad3f076b69d/src/LMIC-node.h#L64

best regards 
CK1

lnlp, 2022-03-13:
Hi,
What is your problem?
Which boardid have you selected in platformio.ini?
What do you want to achieve with LMIC_CONFIG_PRECONDITIONS_H?
(If you read the documentation and check the sources you can see it is defined in platformio.ini)

CK1-CK, 2022-03-14:
Hi,

thanks for answering.

I use the following hardware (because of the current consumption):

- Arduino Pro Mini 3.3 Volt 8 Bit 
- SX1276 Lora Board (Pinmapping with Arduino: .nss = 10, .rst = 9,dio0= 2, dio1= 8)

I just want to monitor a door switch by interrupt PIN (3) and if the pin changes a message should be sent to TTN.

> Apparently you have selected one of the boards with 8-bit AVR microcontroller.
> Can you confirm this?

yes i can confirm

> Which boardid have you selected in platformio.ini?

pro8mhzatmega328 (Arduino Pro Mini 3.3V 8Mhz)

> These 8-bit AVR MCU's are however not preferred for new LoRaWAN developments (due to their limited amount of memory) and neither is the deprecated Classic LMIC library.

-.- :(

I thought i can switch to the "new" MCCI Lib with "#Define _LMIC_CONFIG_PRECONDITIONS_H_ " but than a lot of code errors occur.
```
// Determine which LMIC library is used
#ifdef _LMIC_CONFIG_PRECONDITIONS_H_   
    #define MCCI_LMIC
#else
    #define CLASSIC_LMIC
```

In principle, I don't care if I use an outdated lib. I hope that I can just get a connection with TTN and send the one byte.

Currently the join process fails and no payload will be sent. (see screenshot)
[image]
 At the moment i don't know is it a failure in the arduino code or a wrong config in ttn

[image]

lnlp, 2022-03-14:
> yes i can confirm
>
>
> pro8mhzatmega328 (Arduino Pro Mini 3.3V 8Mhz)

If you read the (LMIC-node) documentation (you should!) you will see that the 8-bit AVR boards have too little memory for LMIC-node in combination with a recent MCCI LMIC library version.

You can still use an 8-bit AVR with MCCI LMIC but you will not have much memory left for more advanced programs.
Try the `ttn-otaa.ino` example included with the MCCI LMIC library together with the MCCI LMIC library. It will also work on 8-bit AVR but is quite a bit more complicated to get it to work than LMIC-node and you will have to avoid any pitfalls yourself.

For LoRaWAN version and regional parameters I usually use version v1.0.3.
I don't remember what the exact differences between v1.0.2 and v1.0.3 are but v1.0.3 usually works for me.

_For more support check The Things Network forum where more people will be able to help._

closed 2022-03-15 by lnlp