lnlp/LMIC-node #37: Event: Uplink not scheduled because TxRx pending  issue

sw-dev-code, 2022-03-31:
I'm using the LMIC-node library on the Heltec LoRa WIFI V2 board and there is the following problem that appears randomly in time:

`Event: Uplink not scheduled because TxRx pending`

After this even if I do rejoin device is unable to send data, only the device reboot helps.

lnlp, 2022-03-31:
Difficult to help find a possible cause without any further details.

Maybe your `DO_WORK_INTERVAL_SECONDS` is set too low?

sw-dev-code, 2022-03-31:
@lnlp I will try to save a log in that scenario. 
DO_WORK_INTERVAL_SECONDS is set to 10.

Understood, the question is more for the MCCI LoRaWAN LMIC library repo.

lnlp, 2022-03-31:
A value of 10 could possibly result in `Uplink not scheduled because TxRx pending`.
If you're testing try a higher value e.g. 30 or 60 (seconds).

For real-life operational/production situations this should normally be much higher.