lnlp/LMIC-node #40: How to change Pico SPI channel?

BNNorman, 2022-04-24:
I have an iLabs Challenger RP2040 Lora (SX1276). 

From what I can tell it uses the second SPI channel to talk to the RFM95W. I have found the pin outs from the schematic for the board (see diagram). That's the relatively easy bit.

How can I tell LMIC-node to use spi1 instead of spi0.

Thanks for any help.

[image]
d

lnlp, 2022-04-25:
LMIC-node uses the LMIC library for LoRaWAN support.
As far as I'm aware LMIC can only use the default SPI port (the instance named SPI).
In order to make LMIC use the second SPI port the pins for the default SPI instance will have to be redefined.

How this can be done is hardware and (in this case MBED) Arduino core dependent.
The ESP32 Arduino core allows SPI pins to be (re)defined via parameters in `SPI.begin()`.
Other cores support methods like `SPI.setMOSI()`, `SPI.setMISO()` etc.
The standard Arduino core for the Pico does not provide any of these.

The IO pin mapping on the Pico appears to be very flexible so there should be some way to redefine the IO pins used for the standard SPI instance but I don't know how. I haven't looked into this.

For an example of how to use different pins for SPI with LMIC-node see BSF `bsf_disco_l072cz_lrwan1.h`.
This uses methods that are not supported by the standard Arduino core for the Pico but it at least gives an example _where_ in LMIC-node code the SPI pins will have to be (re)mapped.

BNNorman, 2022-04-26:
Thank you for your reply. Very helpful. I'll do more digging.

closed 2022-04-26 by BNNorman