lnlp/LMIC-node #41: Failing build: multiple definition of `hal_init'

gkaretka, 2022-04-25:
Hello, I'm building this project with these libraries for `heltec_wireless_stick` board.

Dependency Graph
|-- <U8g2> 2.33.2
|   |-- <SPI> 2.0.0
|   |-- <Wire> 2.0.0
|-- <EasyLed> 1.1.0
|-- <MCCI LoRaWAN LMIC library> 4.0.0
|   |-- <SPI> 2.0.0
|-- <Wire> 2.0.0
|-- <SPI> 2.0.0
Building in release mode

and I am getting this error message
`
Linking .pio\build\heltec_wireless_stick\firmware.elf
c:/users/xxx/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\xxx\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\lib\libpp.a(hal_mac.o): in function `hal_init':
(.text.hal_init+0xc8): multiple definition of `hal_init'; .pio\build\heltec_wireless_stick\libc40\libMCCI LoRaWAN LMIC library.a(hal.cpp.o):C:\Users\xxx\Documents\GitHub\LMIC-node/.pio/libdeps/heltec_wireless_stick/MCCI LoRaWAN LMIC library/src/hal/hal.cpp:416: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\heltec_wireless_stick\firmware.elf] Error 1
`

JohnMacrae, 2022-04-25:
You can solve this by using v3.5.0 of the espressif 32 platform.  Although not a permanent fix, it will get you going. N.B. you will need to uninstall espressif 32 completely and then reinstall 3.5.0. HTH

CaptnAwsome, 2022-04-28:
Hello,

I had the same issue using a `TTGO LoRa32 v2.1` board

It seems to be related to the MCCI LoRaWAN LMIC Library https://github.com/mcci-catena/arduino-lmic and ESP32 based boards.
The same github issue exists on their github page: https://github.com/mcci-catena/arduino-lmic/issues/714
They suggests adding `#define hal_init LMICHAL_init` to the `lmic_project_config.h` or if using PlatformIO to the command line. 
https://github.com/mcci-catena/arduino-lmic/issues/714#issuecomment-822051171

So I got it working by adding  `-D hal_init=LMICHAL_init` to the `mcci_lmic` `build_flags` in the `platformio.ini` file

lnlp, 2022-04-30:
Thanks! @CaptnAwsome and @JohnMacrae 

I have [added the workaround](https://github.com/lnlp/LMIC-node/commit/157754e8164999349c6e7e80bb027a8e310c8ddc) so the issue has been fixed.

closed 2022-04-30 by lnlp