lnlp/LMIC-node #43: USE_DISPLAY Not working TTGO-Lora32 V1.3

kskenyon, 2022-05-20:
I have two new V1.3 TTGO-Lora32 and as far as I can tell everything is configured correctly but the display is either blank or not working.  I know the display is good because it worked with the stock firmware.  Has something changed in recent V1.3 hardware?

kskenyon, 2022-05-21:
After much diddling around I discovered that the new v1.3 marked board uses the code base for v2.  So just select v2 when uploading to this board.  It does not have a controllable LED.

PLAyerNos, 2022-05-22:
Selecting 'heltec_wifi_lora_32' is another option👍

lnlp, 2022-05-26:
@kskenyon 
I just did a brief check in de V1.3's schematic diagram and it appears that definition of SDA and SCL pins in LMIC-node are set incorrectly for the TTGO LoRa32 V1.3. The I2C pin definitions used are (still) from v1.0 versions of this board.

I will have a further look at the issue and add a fix later.

I have none of the TTGO LoRa32 v1.x boards myself for testing. Unfortunately documentation for the TTGO boards is often poor, not available or contains errors, there is little consistency between different versions of their boards and proper Arduino support for each board version is lacking, making it very difficult to support these boards.

julianwue, 2023-02-17:
@lnlp 
**Had the same problem with an old Lora32 v1. The I2C reset pin was missing, but with PIN 16 the display works for me.**

Changed the file "bsf_ttgo_lora32_v1.h" in line 121 from

`U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*rst*/ U8X8_PIN_NONE, /*scl*/ 15, /*sda*/ 4);`

to

`U8X8_SSD1306_128X64_NONAME_HW_I2C display(/*rst*/ 16, /*scl*/ 15, /*sda*/ 4);`

_Also the Board LED works on pin 2 with this code in line 117_ 
`EasyLed led(LED_BUILTIN, EasyLed::ActiveLevel::High);`
_(and uncommented -D USE_LED 423 in platformio.ini )_