lnlp/LMIC-node #45: Bootloop with latest platform-espressif32 (v4.30)

hekopath, 2022-05-26:
I have observed a bootloop after using the latest espressif32 plattform with a ttgo_lora32_v21 board.
My personal workaround was to modify the specify the previous version in the env section for my board variant.

[env:ttgo_lora32_v21]
platform = espressif32@4.2.0

Can anybody confirm this problem?

Tested with the current main branch (v.1.3.0 version fail to build because of #41 )

damanti-me, 2022-06-16:
Thanks for your workaround - i have the same board and get the same issue with latest espressif32.

lnlp, 2022-07-01:
@hekopath @damanti-me 

I only have the V2.0 of this board and none of the V2.1.x versions.

I tried to reproduce the issue with the V2.0 board but was unable to reproduce it.
I also tried with a TTGO T-Beam V0.5 board and both worked fine, without showing the issue.

For the test I upgraded all platforms and libraries (`pio pkg update`). Platform Espressif32 was then updated from v4.2.0 to v5.0.0.
I then removed (`delete`) the platform Epressif32 installation to start with a 'clean' situation.
I replaced all `platform=espressif32` instances in `platformio.ini` with 'platform=espressif32@4.3.0`.
I then selected board-id `ttgo_lora32_v2` and pressed 'Upload'.

This initiated install of the Espressif32 v4.3.0 platform.
The build and upload both succeeded and I was able to run LMIC-node using Espressif32 v4.3.0 on both the TTGO LoRa32 V2.0 and TTGO T-Beam V0.5.

Below output for the TTGO LoRa32 V2.0 while using Espressif32 v4.3.0:

### Build

```text
Processing ttgo_lora32_v2 (platform: espressif32@4.3.0; board: ttgo-lora32-v2; framework: arduino)
------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v2.html
PLATFORM: Espressif 32 (4.3.0) > TTGO LoRa32-OLED V2
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.33.8
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- EasyLed @ 1.1.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|   |-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
...
```

### Running

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

LMIC-node

Device-id:     generic-otaa
LMIC library:  MCCI
Activation:    OTAA
Interval:      60 seconds

000000027871:  Event: EV_JOINING

000000029733:  doWork job started
000000385513:  Event: EV_TXSTART
000000708409:  Event: EV_JOINED
...
```

**So is this bootloop issue with Espressif32 v4.3.0 related specifically to the TTGO LoRa32 V2.1.6 board or could there be other reasons that caused the issue for you?**

By the way, Espressif32 v5.0.0 is already available. Are you still experiencing the same bootloop issue with v5.0.0?
_Don't forget to replace any `platform=espressif32@4.2.0` instances in `platformio.ini` with `platform=espressif32` again before testing._

damanti-me, 2022-07-01:
thank you for your time and investigation. 
I repeated the process with different versions of Espressif32 library. For both (v5.0.0 and v4.4.0) i get the bootloop. And 4.2.0 is working for me. So it seems to be related to Espressif32 lib. If you tell me other ways to verify im would be glad to help.

hekopath, 2022-07-01:
To not alter the environment to much i just upgraded within PlatformIo
```
Platform Manager: Updating espressif32 @ 4.3.0
Platform Manager: Removing espressif32 @ 4.3.0
Platform Manager: espressif32@4.3.0 has been removed!      
Platform Manager: Installing platformio/espressif32 @ 5.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@5.0.0 has been installed!
```

The base of this test is still the main banch from 26th May.
Only modified the lorawan-keys.h
And now with removed @4.2.0 Tag

**Build**
```
Processing ttgo_lora32_v21 (platform: espressif32@4.2.0; board: ttgo-lora32-v21; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.html
PLATFORM: Espressif 32 (4.2.0) > TTGO LoRa32-OLED v2.1.6
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20002.0 (2.0.2)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.33.8
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- EasyLed @ 1.1.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|   |-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\ttgo_lora32_v21\src\LMIC-node.cpp.o
Generating partitions .pio\build\ttgo_lora32_v21\partitions.bin
Compiling .pio\build\ttgo_lora32_v21\lib19a\SPI\SPI.cpp.o
Compiling .pio\build\ttgo_lora32_v21\lib287\Wire\Wire.cpp.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\U8g2lib.cpp.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\U8x8lib.cpp.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\mui.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\mui_u8g2.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_bitmap.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_box.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_buffer.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_button.c.o
Archiving .pio\build\ttgo_lora32_v21\lib19a\libSPI.a
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_circle.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_cleardisplay.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_d_memory.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_d_setup.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_font.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_fonts.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_hvline.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_input_value.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_intersection.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_kerning.c.o
Archiving .pio\build\ttgo_lora32_v21\lib287\libWire.a
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_line.c.o
Compiling .pio\build\ttgo_lora32_v21\libb27\U8g2\clib\u8g2_ll_hvline.c.o
[... 261 lines cut ...]
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57
```

I can conclude the bootloop situation persists.

lnlp, 2022-07-02:
> If you tell me other ways to verify im would be glad to help.

Thanks. To start, providing logging as shown above will be useful.

When _exactly_ does the bootloop occur? Based on @damanti-me's log above the upload appears to succeed after which the device is automatically reset and then right at the first run the boot loop starts. Is that correct?

lnlp, 2022-07-02:
> The base of this test is still the main banch from 26th May

Do you mean LMIC-node's latest commit from 2022-04-30?

lnlp, 2022-07-02:
Regarding:

> ```text
> Platform Manager: Updating espressif32 @ 4.3.0
> Platform Manager: Removing espressif32 @ 4.3.0
> Platform Manager: espressif32@4.3.0 has been removed!      
> Platform Manager: Installing platformio/espressif32 @ 5.0.0
> Downloading  [####################################]  100%
> Unpacking  [####################################]  100%
> Platform Manager: espressif32@5.0.0 has been installed!
> ```
> 
> The base of this test is still the main banch from 26th May.
> Only modified the lorawan-keys.h
> And now with removed @4.2.0 Tag
```

**The following is remarkable:**

```text
> Processing ttgo_lora32_v21 (platform: espressif32@4.2.0; board: ttgo-lora32-v21; framework: arduino)
> -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.html
> PLATFORM: Espressif 32 (4.2.0) > TTGO LoRa32-OLED v2.1.6
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20002.0 (2.0.2)
>  - tool-esptoolpy @ 1.30300.0 (3.3.0)
>  - tool-mkfatfs @ 2.0.1
>  - tool-mklittlefs @ 1.203.210628 (2.3)
>  - tool-mkspiffs @ 2.230.0 (2.30)
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

Above build log shows `espressif32@4.2.0` and `PLATFORM: Espressif 32 (4.2.0)`.
So either the build log you showed is incorrect (from a different build session) or the `@4.2.0` part was not removed everywhere in `platformio.ini`.

**Also remarkable is:**

> PACKAGES:
>  - framework-arduinoespressif32 @ 3.20002.0 (2.0.2)

While my build log _(using Espressif32@4.3.0)_ shows:

> PACKAGES:
> - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)

My build log shows a different (newer) version for framework-arduinoespressif32 (ESP32 Arduino Core).

<br/>
---

So how come your build log shows `PLATFORM: Espressif 32 (4.2.0)` instead of `PLATFORM: Espressif 32 (4.2.0)`?
Is the log incorrect or is there something wrong with your PlatformIO configuration?

Have you tried to re-install PlatformIO from scratch already?
_You can rename your existing `.platformio` folder (or whatever it's called on your OS) first (or make a copy) so you can easily restore it later after testing._
  
<br/>
---

It is strange that our logs shown the same `PLATFORM: Espressif 32 (4.2.0)` but our `framework-arduinoespressif32` versions are different.

I'm not sure what may cause the difference but I would expect that each 'Espressif 32 platform version' comes with it's own specific ESP32 Arduino core (`framework-arduinoespressif32`) version and that when you specify a specific Espressif 32 platform version (e.g. `espressif32@4.2.0`) that it will automatically use the ESP32 Arduino core version that was included with PlatformIO's Espressif 32 platform package.

lnlp, 2022-07-02:
What operating system and version are you using?

TD-er, 2022-07-02:
Also keep an eye on the used esptool and elf2bin versions.

The newer version (4.x) does call elf2bin with parameters to append an SHA checksum of the binary.
But if it was already "signed" with such a checksum, it will not patch the binary when flashing to set the correct flash size and flash clock frequency.
Simply because changing these bits will corrupt the checksum.

The 4.x version of esptool does check for this, but if the bin was made including the SHA and esptool doesn't check for it, it may corrupt the bin while flashing.
On the other hand, of the wrong flash parameters were set in the board definition or PIO env, then esptool cannot correct for it when flashing.

lnlp, 2022-07-02:
Thanks.

So to my understanding: After updating the ESP32 platform version in PlatformIO it will be best to delete the `.pio/build` folder to remove the binary to prevent any possible issue with what you desribed?

TD-er, 2022-07-02:
And also make sure esptool is not mentioned with a specific (different) version in the requirements.txt for the Python virtual env.

lnlp, 2022-07-02:
> It is strange that our logs shown the same `PLATFORM: Espressif 32 (4.2.0)` but our `framework-arduinoespressif32` versions are different.

Silly me, _my_ above logs are from testing with @4.3.0 while your build log shows @4.2.0. This explains the version difference for `framework-arduinoespressif32`.

hekopath, 2022-07-02:
> @damanti-me
> 
> > If you tell me other ways to verify im would be glad to help.
> 
> Thanks. To start, providing logging as shown above for each of the tested configurations will be useful.
> 
> When _exactly_ does the bootloop occur? Based on @hekopath's log above the upload appears to succeed after which the device is automatically reset and then right at the first run the boot loop starts. Is that correct?

Yes!

hekopath, 2022-07-02:
> > The base of this test is still the main banch from 26th May
> 
> Do you mean LMIC-node's latest commit from 2022-04-30?

Yes!

hekopath, 2022-07-02:
> Which operating system and version are you using?

W10 Pro 21H2

hekopath, 2022-07-02:
I tried another build.
With my last build documentation i only used the project task Default->General->"Clean All All"

This time i further cleaned up by deleting .pio and .vscode folder.

Used platformio.ini is the following:

```
;-------------------------------------------------------------------------------`
;
;  File:         platformio.ini
; 
;  Function:     Project configuration file for LMIC-node.
;
;  Copyright:    Copyright (c) 2021 Leonel Lopes Parente
; 
;  License:      MIT License. See accompanying LICENSE file.
;
;  Author:       Leonel Lopes Parente
;
;  Description:  This is the main configuration file. It contains:
;                - Board selector to select your board type
;                - Common settings used for all boards
;                - Board specific settings that can be altered per board.
;                
;                For a description of all settings see README.md
;
;-------------------------------------------------------------------------------

; ------------------------------------------------------------------------------
; |  Board Selector                                                            |
; |                                                                            |
; |  Select your board by uncommenting EXACTLY ONE board-id below.             |
; ------------------------------------------------------------------------------

[platformio]
default_envs = 
    ;<platformio.ini board selector guard> Comment this line and uncomment one board-id below:

    ; LoRa development boards with integrated LoRa support:

    ; Board-id                            Board name
    ;---------                            ----------
    ; adafruit_feather_m0_lora          ; Adafruit Feather M0 LoRa
    ; disco_l072cz_lrwan1               ; Discovery B-L072Z-LRWAN1
    ; heltec_wifi_lora_32_v2            ; Heltec Wifi LoRa 32 V2
    ; heltec_wifi_lora_32               ; Heltec Wifi LoRa 32
    ; heltec_wireless_stick_lite        ; Heltec Wireless Stick Lite
    ; heltec_wireless_stick             ; Heltec Wireless Stick
    ; lopy4                             ; Pycom Lopy4
    ; lora32u4II                        ; BSFrance LoRa32u4 II v1.0, v1.1, v1.2, v1.3
    ; ttgo_lora32_v1                    ; TTGO LoRa32 v1.3
    ; ttgo_lora32_v2                    ; TTGO LoRa32 v2.0
     ttgo_lora32_v21                   ; TTGO LoRa32 v2.1.6
    ; ttgo_t_beam                       ; TTGO T-Beam v0.5, v0.6, v0.7
    ; ttgo_t_beam_v1                    ; TTGO T-Beam v1.0, v1.1

    ; Development boards that require an external SPI LoRa module:

    ; Board-id                            Board name
    ;---------                            ----------
    ; adafruit_qt_py_m0                 ; Adafruit QT Py    
    ; blackpill_f103c8_128k             ; Black Pill 128k
    ; blackpill_f103c8                  ; Black Pill  64k
    ; bluepill_f103c8_128k              ; Blue Pill 128k
    ; bluepill_f103c8                   ; Blue Pill  64k
    ; lolin_d32_pro                     ; Lolin D32 Pro
    ; lolin_d32                         ; Lolin D32
    ; lolin32                           ; Lolin32
    ; nodemcu_32s                       ; NodeMCU-32S
[... 1129 lines cut ...]
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4

--- exit ---
```

hekopath, 2022-07-02:
As an addition:
Adding the "@4.2.0" Tag immidiatly result in a successful running build even without cleanups.

**Build, Upload and Running**
```
> Executing task: C:\Users\hendrik\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor <

Processing ttgo_lora32_v21 (platform: espressif32@4.2.0; board: ttgo-lora32-v21; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-lora32-v21.html
PLATFORM: Espressif 32 (4.2.0) > TTGO LoRa32-OLED v2.1.6
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20002.0 (2.0.2)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.33.8
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- EasyLed @ 1.1.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|   |-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\ttgo_lora32_v21\firmware.elf
Checking size .pio\build\ttgo_lora32_v21\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.7% (used 18660 bytes from 327680 bytes)
Flash: [==        ]  21.6% (used 283113 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM16
Uploading .pio\build\ttgo_lora32_v21\firmware.bin
esptool.py v3.3
Serial port COM16
Connecting.......
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 50:02:91:9d:da:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00055fff...
Compressed 17088 bytes to 11814...
Writing at 0x00001000... (100 %)
Wrote 17088 bytes (11814 compressed) at 0x00001000 in 0.4 seconds (effective 313.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 346.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 580.0 kbit/s)...
Hash of data verified.
Compressed 283504 bytes to 162243...
Writing at 0x00010000... (10 %)
Writing at 0x0001999e... (20 %)
Writing at 0x000234bf... (30 %)
Writing at 0x00028733... (40 %)
Writing at 0x0002dbd3... (50 %)
Writing at 0x000332aa... (60 %)
Writing at 0x000394d0... (70 %)
Writing at 0x000439d0... (80 %)
Writing at 0x0004ad70... (90 %)
Writing at 0x000501ac... (100 %)
Wrote 283504 bytes (162243 compressed) at 0x00010000 in 2.7 seconds (effective 842.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================================================== [SUCCESS] Took 11.24 seconds ======================================================================================
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM16  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4

LMIC-node

Device-id:     ttgo-lora32-v21
LMIC library:  MCCI
Activation:    OTAA
Interval:      60 seconds

000000027816:  Event: EV_JOINING

000000029740:  doWork job started
000000183762:  Event: EV_TXSTART 
```

lnlp, 2022-07-03:
Observations so far:
- Issue is Espressif32 platform related. Version v4.3.0 and newer cause the issue. Versions up until v4.2.0 work fine.
- Espressif32 platfrom v5.0.0 does not fix it and still causes the issue.
- Issue appears TTGO LoRa32 V2.1.6 _(V2.1.x)_ specific. 
_I was unable to reproduce it for TTGO LoRa32 V2.0 and TTGO T-Beam V0.5._

I don't have time to dive into this issue any further currently and neither have the TTGO LoRa32 V2.1.6 hardware to test it.

Next steps for anyone wanting to further investigate the issue could be to do a file diff analysis between Espressif32 platform versions  v4.2.0 and v4.3.0 and make a diff analysis between their respective `framework-arduinoespressif32` package versions and/or contact PlatformIO.

lnlp, 2022-07-03:
@hekopath @damanti-me 

Do you have any of the other supported ESP32 based LoRa boards that you could test with Espressif32 platform v4.3.0(+)?
Have you tried to upload standard Arduino sketches (e.g. blink) with PlatformIO to your board to see if the issue persists or not? That way you can test if the issue is more general or only occurs with LMIC(-node)?

TD-er, 2022-07-03:
Check the board definitions to see what's different.
Especially the flash settings.
You can also override those in the platform.ini file

hekopath, 2022-07-03:
> @hekopath, @damanti-me
> 
> Do you have any of the other supported ESP32 based LoRa boards that you could test with Espressif32 platform v4.3.0(+)?

I have now tested it with a ttgo-lora32-v1 without any issues.

> 
> Have you tried to upload standard Arduino sketches (e.g. blink) with PlatformIO to your board to see if the issue persists or not? That way you can check if the issue is more general or only occurs with LMIC(-node).

Furthermore i tried to flash cyberman54/ESP32-Paxcounter with a similar list of supported boards and can confirm the same behavior as with LMIC-node .

ttgo-lora32-v1 build, upload and running without loop
ttgo-lora32-v21 (ttgo-lora32-v21new on paxcounter) results in a bootloop without the version tag @4.2.0

So i can confirm that it is not a LMIC-node specific problem.
Sorry for bothering you with this and many thanks for assisting to narrow the issue down a bit.

At least it is usefull to raise this issue  and provide a workaround as @damanti-me did.

lnlp, 2022-07-03:
> So i can confirm that it is not a LMIC-node specific problem.

Good to have this clear now. 🙂

It will still be good to know if possibly LMIC library related or not.
**So could you please also test with something simple like the standard **`Blink`** sketch?**
If the issue also manifests with something simple like `Blink` then it will be easier to reproduce which may make it easier to get this fixed by PlatformIO or Espressif..

<br/>

> Sorry for bothering you with this and many thanks for assisting to narrow the issue down a bit.
> At least it is usefull to raise this issue and provide a workaround as @damanti-me did.

You're welcome. Others have ran (and will run) into this issue too, so it's good there is a workaround available.

I will leave this issue open for now so others can more easily find it.

lnlp, 2022-07-03:
# Workaround

As already described by @hekopath the workaround can be implemented as follows:

In `platformio.ini` replace

```ini
[env:ttgo_lora32_v21]
; TTGO LoRa32 v2.1.6 (ESP32).
; Onboard OLED display SSD1306 0.96" 128x64.
platform = espressif32
```

with

```ini
[env:ttgo_lora32_v21]
; TTGO LoRa32 v2.1.6 (ESP32).
; Onboard OLED display SSD1306 0.96" 128x64.
platform = espressif32@4.2.0
```

<br/>

_Once the issue gets fixed in a newer version of Espressif32 platform, above workaround should be undone again._

ezCheater, 2022-09-05:
I had the same Issue with my TTGO LoRa32 V2.1.6 board. I was able to "fix" the bootloop by changing the value in pins_arduino.h form
```
#define OLED_RS 16
```
to
```
#define OLED_RS -1
```
I can't really explain why it works, but my collague, who suggested the fix, said something along the lines of cheaper OLED Displays dont connect the reset Pin, which causes these kinds of Problems.

TD-er, 2022-09-05:
GPIO-16 and -17 might also be connected to SPI flash or PSRAM.
Even when no PSRAM is present, the pin might be involved in probing for PSRAM and thus it might affect flash IO when you change its state.

alexschultze, 2023-03-31:
I had the same issue on a cheap "LILYGO TTGO" branded board.
OLED_RST -1 solved the issue.