lnlp/LMIC-node #49: Add support for heltec LoraWAN Wifi OLED board version 3

smbunn, 2022-11-28:
Works great with v2 boards, can we have a v3 update?

PProvost, 2023-02-21:
My understanding is that this will be waiting for the PR that has ESP32 S3 support to make it to the arduino-esp32 project. I ran into this a while ago when I ordered a v2 and got a v3. Now it sits on my bench until the code catches up.

c64emulator, 2023-04-19:
As already stated this repo provides support HELTEC LoRa32 **v2**. This device is EOL.
The new **v3** device is out since several month with other CPU (ESP32-S3) and other LoRa chip (SX1262) . The LMIC-lib  seems not to work with this device because lacking support for SX1262.
The Heltec WiFi Kit Series (V0.0.7) [https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/releases/tag/0.0.7] (Basetype: LoRaMac-node) is (currently) the only working environment.
PLS provide support for the new version of the HELTEC LoRa32 board. Thank you.

nikito7, 2023-10-02:
> The LMIC-lib seems not to work with this device because lacking support for SX1262.

It's the real problem...