lnlp/LMIC-node #50: rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

MrAnnihilaToR, 2023-01-06:
Hello guys, I have this issue on my LilyGo TTGO v2.1.6 ESP32 LoRa32 - after I click the reset button to start, it gets in an infinite bootloop , probably some watchdog or something ... :

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
ets Jun  8 2016 00:22:57

If anyone knows how to fix it, I'll be very thankful :)

vkynchev, 2023-01-06:
duplicate of #45

MrAnnihilaToR, 2023-01-06:
> duplicate of #45

Thanks, man, appreciated! :)

closed 2023-01-06 by m-andreev

LucaH360, 2023-01-14:
j'essaye de faire marche une caméra ESP 32 webserveur mais je dois télécharger ça mais je saos pas comment merci d'avence : DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2