lnlp/LMIC-node #53: Error ESP32s + Lilygo T-Highrow lora shield 20191223

0d3ng, 2023-02-22:
Hello everyone,
Anyone have an error like the one below
`LMIC-node

Device-id:     nodemcu-32s
LMIC library:  MCCI
Activation:    OTAA
Interval:      60 seconds

FAILURE
.pio/libdeps/nodemcu_32s/MCCI LoRaWAN LMIC library/src/lmic/oslmic.c:53`

i used wiring like below
SPI/LoRa module     GPIO
 *                ---                 ----
 *                MOSI  <――――――――――>  23  (MOSI)
 *                MISO  <――――――――――>  19  (MISO)
 *                SCK   <――――――――――>  18  (SCK)
 *                NSS   <――――――――――>   5  (SS)
 *                RST   <――――――――――>  27
 *                DIO0(IO14)  <――――――――――>  34
 *                DIO1(IO13)  <――――――――――>  35
 *                DIO2                 -  Not needed for LoRa.

When i compiled the log message
`PLATFORM: Espressif 32 (4.2.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20002.220503 (2.0.2)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.34.13
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- EasyLed @ 1.1.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|   |-- SPI @ 2.0.0
|-- Wire @ 2.0.0`

Thanks for your help

gitHelmut, 2023-03-03:
Look into platformio.ini and change bold line:

[mcci_lmic]
; LMIC-node was tested with MCCI LoRaWAN LMIC library v3.3.0 and v4.0.0.
; Some changes have been announced for future versions of the MCCI library
; which may be incompatible with LMIC-node. In case of problems just
; use mcci-catena/MCCI LoRaWAN LMIC library@4.0.0 below which will
; explicitly use v4.0.0 of the library.
; Perform 'PlatformIO: Clean' after changing library version and
; in case of issues remove the old version from .pio/libdeps/*.

; Note: LMIC_PRINTF_TO is defined for each board separately
;       in the board specific sections. Don't define it in this section.

lib_deps =
    ; Only ONE of below LMIC libraries should be enabled.
    ; mcci-catena/MCCI LoRaWAN LMIC library           ; MCCI LMIC library (latest release)
    **mcci-catena/MCCI LoRaWAN LMIC library@4.0.0   ; MCCI LMIC library v4.0.0**

jeroenrnl, 2023-10-25:
I had the same problem. Swap pin 5 and pin 18...