lnlp/LMIC-node #56: Doesn't build with target platform set to 'nodemcu2'

virtyaluk, 2023-07-19:
The build log:
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- U8g2 @ 2.35.3
|-- EasyLed @ 1.1.0
|-- MCCI LoRaWAN LMIC library @ 4.1.1
|-- Wire @ 1.0
Building in release mode
Compiling .pio/build/nodemcuv2/src/LMIC-node.cpp.o
In file included from src/LMIC-node.h:82,
                 from src/LMIC-node.cpp:53:
src/boards/bsf_nodemcuv2.h:68:10: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://registry.platformio.org/search?q=header:ESP8266WiFi.h
*
*********************************************************************

   68 | #include <ESP8266WiFi.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1608.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1609.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1610.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1611.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1617.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1638.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1701_dogs102.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_d_uc1701_mini12864.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_debounce.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_display.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_fonts.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_gpio.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_input_value.c.o
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_message.c.o
*** [.pio/build/nodemcuv2/src/LMIC-node.cpp.o] Error 1
Compiling .pio/build/nodemcuv2/libba3/U8g2/clib/u8x8_selection_list.c.o
=========================================================== [FAILED] Took 1.93 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:01.927
====================================================== 1 failed, 0 succeeded in 00:00:01.927 ======================================================
```

It works just fine for other platforms like `pico`:

[image]

virtyaluk, 2023-07-19:
Removing `#include <ESP8266WiFi.h>` from `./src/boards/bsf_nodemcu2.h` turned the build green though.

nikito7, 2023-10-02:
Or add ESP8266WiFi to lib_deps.

You may need mac address or force wifi off.