lnlp/LMIC-node #58: Wrong Pi Pico pinout

jarleven, 2024-01-24:
I think the pin mapping for the Pi Pico is incorrect.
Managed to figure out the pinout and got it working with the RFM95W. 
[image]

I believe this pin mapping is correct:

```
 *                SCK   <――――――――――>  18   (SCK)      (PIN_SPI_SCK)
 *                MOSI  <――――――――――>  19   (MOSI)     (PIN_SPI_MOSI)
 *                MISO  <――――――――――>  16   (MISO)     (PIN_SPI_MISO)
 *                NSS   <――――――――――>  17   (SS)       (PIN_SPI_SS)
 ```

s00500, 2024-07-04:
THANKS SO MUCH, I can very much confirm that this is an issue

s00500, 2024-07-04:
(and that it works fine with the pinout you are documenting here)